mongoose-os-libs/wifi #5: WiFi Event when station is connected to AP

mglettig, 2018-03-05:
The Espressif SDK triggers an event when a station is connected to the Access-Point and when the station is disconnected. On the following picture those events are highlighted (source: https://github.com/espressif/ESP8266_NONOS_SDK):
[image]

Please dispatch those events to the "mgos_wifi_changed_t" callback:
- MGOS_WIFI_AP_DISCONNECT
- MGOS_WIFI_AP_CONNECT

Or is there another way to figure out if a station is connected when the ESP is in AP-mode?

mglettig, 2018-04-10:
@rojer The PR that I created some time ago contains those new events. Could you incorporate this changes? Thank you for the feedback.

rojer, 2018-04-10:
yeah, i've had a quick glance at it before but it's not quite there yet. i support the ide in principle, but we need to apply it to all our platforms. this will need more work.

tripflex, 2018-05-19:
👍

junym, 2018-06-15:
Hi @mglettig, could you point me where is this implementation  and how I can use?

MopeZe, 2018-06-15:
Hi @junym 
You can find it here: https://github.com/mongoose-os-libs/wifi/pull/7/commits/9a1217a7c51a36b9c4d1fc980b1cfe1006c8b4e0

junym, 2018-06-15:
Thanks @MopeZe!   Sorry my foolishness, but whats I should to make for use this changes.  In my  mean mos.yml I should to add this path < https://github.com/mongoose-os-libs/wifi/tree/9a1217a7c51a36b9c4d1fc980b1cfe1006c8b4e0 > ?  

A another thing that I can't understanding is if I can point a origin libs to my own libs in github, cause when I make this I get compilation errors like this : 

Error 1
/mongoose-os/fw/src/mgos_config.mk:14: recipe for target '/fwbuild-volumes/2.2/apps/wifi-setup-web/esp8266/build_contexts/build_ctx_827088842/build/gen/mgos_config.c' failed

tripflex, 2018-06-15:
Yeah this is pretty important to be able to know when someone connects to the AP .. unless there's another way you guys can point out?  @rojer even if it can't be added for the other platforms outside of the Espressif SDK, could you not just add the event (or some kind of generic way) using the Espressif c files (or even just check if platform is espressif before triggering event, and do nothing for other platforms if no way exist with them) .... feels like other platforms would be holding this back on ones that support it??

andyjjones1, 2018-07-30:
Are we able to know if someone connects to the AP?  Is there some kind of event I can hook into?

rojer, 2018-07-30:
@andyjjones1 not at the moment. but watch this space.

closed 2018-08-01 by rojer