mongoose-os-libs/wifi #8: Crashes on Enterprise Authentication

scaprile, 2018-05-18:
Please see forum, starting here:
https://forum.mongoose-os.com/discussion/comment/6691/#Comment_6691

the running application has minimal JS code and runs stable on non-Enterprise WiFi

scaprile, 2018-06-07:
Analyzing core dumps I get:

- First crash (both core dumps):
```
(gdb) bt
#0  0x4000c2e0 in memcpy ()
#1  0x4018de90 in wpa_set_pmk ()
#2  0x4019027d in wpa2Task ()
```
- Second crash (both core dumps):
```
(gdb) bt
#0  0x400014fd in strlen ()
#1  0x400f1270 in mgos_wifi_dev_sta_setup (
    cfg=0x3ffd0500 <mgos_sys_config+472>)
    at /fwbuild-volumes/2.2/apps/PSA/esp32/build_contexts/build_ctx_584064721/libs/wifi/esp32/src/esp32_wifi.c:298
#2  0x400f075c in mgos_wifi_setup_sta (cfg=0x3ffd0500 <mgos_sys_config+472>)
    at /fwbuild-volumes/2.2/apps/PSA/esp32/build_contexts/build_ctx_584064721/libs/wifi/src/mgos_wifi.c:239
#3  0x400f09b1 in mgos_wifi_setup (cfg=0x3ffd04c4 <mgos_sys_config+412>)
    at /fwbuild-volumes/2.2/apps/PSA/esp32/build_contexts/build_ctx_584064721/libs/wifi/src/mgos_wifi.c:420
#4  0x400f0a9c in mgos_wifi_init ()
    at /fwbuild-volumes/2.2/apps/PSA/esp32/build_contexts/build_ctx_584064721/libs/wifi/src/mgos_wifi.c:480
#5  0x400e94ab in mgos_deps_init ()
    at /fwbuild-volumes/2.2/apps/PSA/esp32/build_contexts/build_ctx_584064721/build/gen/mgos_deps_init.c:345
#6  0x400e2bbf in mgos_init () at /mongoose-os/fw/src/mgos_init.c:56
#7  0x400e28d3 in mgos_init2 () at /mongoose-os/fw/src/mgos_hal_freertos.c:195
#8  0x40082d3b in mgos_task (arg=0x0)
    at /mongoose-os/fw/src/mgos_hal_freertos.c:206
```
* Third crash
  * First dump
```
(gdb) bt
#0  0x4000c310 in memcpy ()
#1  0x401901b3 in wpa2Task ()
```
* Third crash
  * Second dump
```
(gdb) bt
#0  0x4009149b in invoke_abort ()
    at /opt/Espressif/esp-idf/components/esp32/panic.c:139
#1  0x4009159e in abort ()
    at /opt/Espressif/esp-idf/components/esp32/panic.c:148
#2  0x400d12cb in task_wdt_isr (arg=<optimized out>)
    at /opt/Espressif/esp-idf/components/esp32/task_wdt.c:163
#3  0x40081671 in _xt_lowint1 ()
    at /opt/Espressif/esp-idf/components/freertos/xtensa_vectors.S:1105
```
So...
- Should I learn chinese for crashes 1 and 3 ?
- Can Cesanta take care of crash 2 ?
- Is there anything else you need to process this issue ?
- app is attached in case it is of any use
[PSA-esp32-1.0.zip](https://github.com/mongoose-os-libs/wifi/files/2081664/PSA-esp32-1.0.zip)

rojer, 2018-06-08:
ok, looks like we need to check for `wifi.sta.username`. i'll send a patch to fix issue 2

scaprile, 2018-06-11:
@rojer : Thanks, I contacted Espressif (we are local distributors) and they are asking me for "the log". Is there a chance to get the IDF functions log using mos tool ?

scaprile, 2018-06-11:
Regarding crash 3, I could not reproduce it on the IDF tool chain.
Looking closely at the second dump, I see in the console log:
```
[May 15 17:47:11.646] Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[May 15 17:47:11.669]  - mgos (CPU 0/1), backtrace: 0x4000bff0 0x4008f7e9 0x4008ec15 0x4008ed02 0x40081f52 0x40082023 0x401b6a81 0x401b6a95 0x400e7ddf 0x400e0b20 0x40082d8e
[May 15 17:47:11.669] 
[May 15 17:47:11.669] Tasks currently running:
[May 15 17:47:11.669] CPU 0: wifi
[May 15 17:47:11.669] Aborting.
[May 15 17:47:11.669] abort() was called at PC 0x400d12c8 on core 0
[May 15 17:47:11.669] 
[May 15 17:47:11.669] Backtrace: 0x4009149b 0x4009159b 0x400d12c8 0x4008166e 0x400e6b22
[May 15 17:47:11.669] --- BEGIN CORE DUMP ---
```
Could you guys take a look at it ?