mongoose-os-libs/wifi #9: AP disabled/unavailable when trying to connect to incorrect SSID

WooWapDaBug, 2018-05-21:
If you set an incorrect SSID (meaning it can't connect), the AP gets completely disabled (this behaviour has been tested on ESP8266 and ESP32).

This is what console shows over and over again:

```
[May 21 14:46:22.168] no vodafone8058-5G found, reconnect after 1s
[May 21 14:46:22.172] mgos_net_on_change_c WiFi STA: disconnected
[May 21 14:46:22.176] mgos_net_on_change_c WiFi STA: connecting
[May 21 14:46:25.020] scandone
[May 21 14:46:25.022] no vodafone8058-5G found, reconnect after 1s
[May 21 14:46:25.026] mgos_net_on_change_c WiFi STA: disconnected
[May 21 14:46:25.030] mgos_net_on_change_c WiFi STA: connecting
[May 21 14:46:27.873] scandone
[May 21 14:46:27.874] no vodafone8058-5G found, reconnect after 1s
[May 21 14:46:27.878] mgos_net_on_change_c WiFi STA: disconnected
[May 21 14:46:27.883] mgos_net_on_change_c WiFi STA: connecting
```

If you set a correct SSID, both the STA and AP start fine.

My guess is that the recconection is somehow driving the AP crazy (I sometimes see the Wifi AP appear when I do a scan from my phone, then it suddently dissapears and so on).
Is there any way to fix it?
Can the reconnect time be changed?

rojer, 2018-06-06:
true. looks like AP is unavailable while STA is connecting, and since we try to connect all the time, this causes AP to be unavailable all the time.
i am going to send a PR that reduces unavailability to a couple seconds every 30 seconds (`sta_connect_timeout`).

closed 2018-06-06 by cpq

WooWapDaBug, 2018-06-07:
Great! So is this now solved? @rojer

WooWapDaBug, 2018-06-11:
It works. But it doesn't connect to the STA with AP is configured automatically, I need to trigger the connection manually.

junym, 2018-06-13:
Hi guys , I'm reopening this discussion because I tested to change sta_connect_timeout parameter but I can't my esp8266 is trying to reconnect each 1s yet.  I'm using mongoose-os to flashing and I changed this parameter in my mos.yml file.
 So, when it's in AP mode, I can see at webpage the json configuration and notice this parameter (sta_connect_timeout) is really  with 30, but I can't understanding why it's taking   1s to reconnect.

rojer, 2018-06-14:
@junym 

>  my esp8266 is trying to reconnect each 1s

this means you are using old version of this library, prior to the latest change.

junym, 2018-06-14:
Thanks for your response @rojer.  
Ok, I think that  typing  "mos update" in terminal console and  git clone the exemples newly would solves my problem, but I take 1s for reconnect yet. 

[Jun 14 16:31:01.539] no junior_machado found, reconnect after 1s
[Jun 14 16:31:01.546] mgos_net_on_change_c WiFi STA: disconnected
[Jun 14 16:31:01.549] mgos_net_on_change_c WiFi STA: connecting

Maybe I has make anything wrong updating