mongoose-os-libs/wifi #10: Doesn't connect to STA during startup when AP is configured

WooWapDaBug, 2018-06-11:
My mos.ylm is:

`
config_schema:
      - ["wifi.sta.enable", true]

      - ["wifi.ap.enable", true]
      - ["wifi.ap.ssid", "ecogarden"]
      - ["wifi.ap.pass", ""]
` 

I set the ssid and password and I save it in the conf9.json
When I restart it doesn't connect to the sta. But if I force to connect it does connect:

`
    struct mgos_config_wifi_sta sta_cfg;
    memcpy(&sta_cfg, mgos_sys_config_get_wifi_sta(), sizeof(sta_cfg));
    sta_cfg.enable = 1;
    if (!mgos_wifi_setup_sta(&sta_cfg))
    {
        LOG(LL_ERROR, ("Wifi STA setup failed"));
    }
`

Why doesn't it connect during startup?

proceptdev, 2018-12-12:
Does the memcpy() require a free()?

WooWapDaBug, 2019-01-25:
@proceptdev Good question. I have no clue, I used that code from their website. I don't know when that config is used so I wasn't confident to free that memory. Good point.

Anyway, I think this was somehow fix at some point as I don't have this problem any more.

closed 2019-01-25 by rojer