mongoose-os-libs/wifi #16: Debug exception reason: Stack canary watchpoint triggered (wifi)

Tobix99, 2018-08-10:
## Expected Behavior
Connect to my configured wifi station(in mos.yml)

## Actual Behavior
Crashes sometimes on startup with 

`Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (wifi)`

and makes a `CORE DUMP`

I've read, that maybe the PSU is too weak too power the ESP32 when connection to WIFI(peakpower), but I'm running from a Anker 60W 10 Port Hub and think, that it should be sufficient power. 

And when you look at the log, you can see, that after the automatic reboot the system boots fine, so I don't know what has caused the problem. The app is an empty app, only wifi and dash configured.

## Specifications

  - Version: 2.4.3
  - Platform: ESP32
  - Subsystem: NodeMCU

## Log
[log.txt](https://github.com/mongoose-os-libs/wifi/files/2277780/log.txt)

rojer, 2019-05-10:
doesn't seem to happen (anymore), please reopen if it's still a problem.

closed 2019-05-10 by rojer

andrey-kruglikov, 2019-12-04:
@Tobix99:   Hi... I faced to problem looks like yours..... except cernel panic appeares in case internet connection lost and  device trying to reconnect wifi couple of times.
Could you please describe how does it fixed in your case? Thx in advance.

bmedici, 2023-08-11:
Same exact behaviour here: `Debug exception reason: Stack canary watchpoint triggered (wifi)`
Could it be related to one GPIO used and which should not be used with Wifi ?