mongoose-os-libs/wifi #17: How to handle `SYSTEM_EVENT_AP_PROBEREQRECVED` wifi event ?

muratsplat, 2018-09-03:
Hello 

I need to handle `SYSTEM_EVENT_AP_PROBEREQRECVED` event provided wifi driver.

Can you add handling "SYSTEM_EVENT_AP_PROBEREQRECVED" for ESP32/ESP8266 support as new feature ?

rojer, 2018-09-03:
events not handled by the wifi driver fall through to [esp32_system_event_handler_default](https://github.com/cesanta/mongoose-os/blob/e231774cc5191b7775f52cfb7e75ac85fb7e20e8/fw/platforms/esp32/src/esp32_main.c#L57).
the default implementation is empty and marked weak, so all you need to do is define your own function with that name and you will receive PROBEREQ events.

closed 2018-09-03 by rojer

muratsplat, 2018-09-07:
@rojer  Is it possible JS runtime ? Can you share me  a code sample ?