mongoose-os-libs/wifi #38: Mesh networking support

Harvie, 2020-11-19:
Can you please add support for wifi mesh networking?

rojer, 2020-11-19:
what does this mean in practice? can you give some links for reading?

Harvie, 2020-11-19:
You provide all devices with some shared secret and based on that, they all assemble into network in such a way that every node can communicate with every other node even when that means some of the traffic has to be routed through nodes with better signal.

I beleive it's been already supported in esp-idf for years:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/mesh.html
https://www.espressif.com/en/products/sdks/esp-wifi-mesh/overview

Harvie, 2020-11-19:
What is really interresting is fact that one of the devices can connect to regular wifi AP and forward traffic between the mesh network and the AP, so it can basicaly autoconfigure to extend the range of AP to mongoose devices which are too far from AP to get usable signal...

rojer, 2020-11-21:
might be interesting to look into, but unlikely to happen any time soon.

pradeep538, 2023-09-14:
Any progress on this ?